EFeru/hoverboard-sideboard-hack-GD #1: getting Cortex M3 legacy target in uVision

bionic24, 2020-02-10:
Here is a picture of the Windgoo sensorboard that was inside my Hoverboard. The SWD programming pins are directly below the connector for the mainboard. I have measured the pins and added them to the picture. 
[image]

Connecting the board through ST-Link utility and unlocking the flash went okay. No errors.

After that I tried to install uVision and open the .UVPROJ file but I get a lot of errors about missing files for the Genefar Dynamics M3 Cortex boards. I have tried to install the legacy database but not sure what to do next. 

The Hex file that is already in the build folder of the repository doesn't flash with ST-Link utility. Is it possible to get a pre-build bin file of the Hoverboard version?

EmanuelFeru, 2020-02-10:
Wow, your board looks different but maybe the input is the same.

Does it have the same GD32 MCU as in my repo. I am surprised  it doesn't compile the project. It should be all contained...  try  installing the GD32F130 family of devices.
I will check if I can generate a bin  file from my uVision.

bionic24, 2020-02-10:
Chipset appears to be the same GD32F130 and MPU6052.

The problem is probably due to me having missing libraries or wrong settings. I have never used uVision up until now and tbh it is kind of a pain in the behind to install Keil MDK and spending hours on getting a bin file without succes. 

[image]

EmanuelFeru, 2020-02-11:
I managed to generate the .bin file with the steps from [here](http://www.keil.com/support/docs/3213.htm). Try to see if you can flash. You can find the .bin file here:
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/tree/master/build/VARIANT_DEBUG
_______________

Regarding Keil, I installed Keil some time ago and I followed these steps:
https://www.rcgroups.com/forums/showthread.php?2512604-Eachine-H8-mini-acro-firmware

Additionally, I installed the AddOns, in my repo here:
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/master/docs/GD32F1x0_Addon_V3.1.0.rar

If you have time, you can try it. Eventualy these steps will help others as well. I will uptade the ReadMe once we have the steps sorted-out.
If still no success, I will install Keil on my wife's PC which is programming tools free :) to get the installation steps.
______________

bionic24, 2020-02-11:
I think this might be thing with my board and not with the file. I can remove the protection, I can read the chip, I can erase the data. But upon writing I get a constant error on 0X8000000. 

12:38:09 : STM32CubeProgrammer API v2.3.0
12:38:12 : ST-LINK SN : 54FF6C064975525024412467
12:38:12 : ST-LINK FW : V2J34S7
12:38:12 : Voltage : 3.26V
12:38:12 : SWD freq : 4000 KHz
12:38:12 : Connect mode: Normal
12:38:12 : Reset mode : Software reset
12:38:12 : Device ID : 0x410
12:38:13 : UPLOADING OPTION BYTES DATA ...
12:38:13 : Bank : 0x00
12:38:13 : Address : 0x4002201c
12:38:13 : Size : 8 Bytes
12:38:13 : Bank : 0x01
12:38:13 : Address : 0x1ffff800
12:38:13 : Size : 16 Bytes
12:38:13 : UPLOADING ...
12:38:13 : Size : 1024 Bytes
12:38:13 : Address : 0x8000000
12:38:13 : Read progress:
12:38:13 : Data read successfully
12:38:13 : Time elapsed during the read operation is: 00:00:00.008
12:38:35 : Memory Programming ...
12:38:35 : Opening and parsing file: debugtestsideboard11feb.bin
12:38:35 : File : debugtestsideboard11feb.bin
12:38:35 : Size : 87153 Bytes
12:38:35 : Address : 0x08000000
12:38:35 : Erasing memory corresponding to segment 0:
12:38:35 : Erasing internal memory sectors [0 85]
12:38:35 : Download in Progress:
12:38:36 : Error: failed to download Segment[0]
12:38:36 : Error: failed to download the File
12:38:36 : Time elapsed during download operation: 00:00:01.034

EmanuelFeru, 2020-02-11:
Can you try locking and then unlocking again the GD32? I had some strange behavior with one board and did the lock/unlocking with all the "Use configuration option bytes" checked (see [here](https://raw.githubusercontent.com/EmanuelFeru/hoverboard-firmware-hack-FOC/master/docs/pictures/unlock_mcu_3.png), just check all boxes, leave the Pages unchecked).

Then it worked well. In my case I could flash it but it was behaving strange. Locking/Unlocking did the job.

EmanuelFeru, 2020-02-12:
@bionic24 I have the same Error if I try to flash with ST Link Utility. Apparently is not able to flash the device.
Finally, I figured out how to build with Platformio see [build-gd32-project-with-platformio](https://community.platformio.org/t/build-gd32-project-with-platformio/11944)
The guys from Platformio were extremely helpful and now I have a project that runs with Platformio too. I will commit soon. And also describe the steps for Keil uVision.

EmanuelFeru, 2020-02-13:
Platformio committed, check the steps in ReadMe.
Other than that, I installed Keil uVision on a fresh PC and loaded the project. After accepting the installation of the missing packages, the project built without any issues, 0 Errors, 0 Warning. So, for me Keil worked without any hiccups.

bionic24, 2020-02-14:
Succes!

Platformio opens the project and makes the builds without errors. The original sideboard from the first post still gave me the same issue. After that I tried the right side sideboard. It is equipped with a BT module but the same otherwise. That board was able download the builds. So I probably have a hardware failure in the first board.

Hope to get some testing done this weekend. I probably need to disable the steering value in the FOC firmware to disable it? I have both motors bolted together in the rim of my Onewheel so it is mandatory that they operate exactly the same.
[image]
[image]

EmanuelFeru, 2020-02-14:
Yes, you will need to disable Steering, check the HOVERCAR variant to see how is done.

closed 2020-02-16 by bionic24