EFeru/hoverboard-sideboard-hack-GD #2: What does it do?

gnbl, 2020-02-19:
Hi, could you add an abstract to the readme, pointing out what this firmware does, e.g. does it reimplement the original behavior? Thanks.

closed 2020-02-19 by gnbl

EmanuelFeru, 2020-02-19:
Hi,  I coded this firmware from scratch, so it does not replicate the original behavior. 
What I try to do is to make here is an easy to understand framework, on which people can work on, for projects like hoverboard, but not only, also other projects that need IMU, leds, photosensors.. etc... 
The sideboard is actually a Blue pill with more hardware.

Depending on the time, I will try to restore the hoverboard functionality in the future, so combing this branch with the FOC firmware for the mainboard.

gnbl, 2020-02-19:
Sorry, I saw https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD#example-variants too late.
Thanks for sharing this, it might come in handy as sensor board for an ESC (S for stability) for a "hovercar"...