EFeru/hoverboard-sideboard-hack-GD #3: GD32F130C8T6

tonvanhieu, 2020-02-25:
My sideboard chipset is GD32F130C8T6 different from yours (C6T6). I have flashed your firm ware to my chip success but it can't read MPU data. Do you know  what i  have to do to make project is suitable with C8T6 chip

EmanuelFeru, 2020-02-25:
Looking at the memory map [GD32F130C8T6](https://datasheet4u.com/datasheet-parts/GD32F130C8T6-datasheet.php?id=976683) vs [STM32F103C8T6](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-STM/blob/master/docs/stm32f103c8-Datasheet.pdf)  there are similarities but also differences. So, I don't think the [hoverboard-sideboard-hack-STM](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-STM) version would match.

I think you need to download the `GD32F1x0_Firmware_Library_V3.1.0` from here:
https://gd32mcu.21ic.com/documents
search for `GD32F1x0` and download the files. Then replace the library.

In quick words this should be the steps. Let me know about the progress.

[image]

EmanuelFeru, 2020-02-25:
oh, I just noticed is the same GD library, the C8T6 has only more memory than the C6T6.
After flashing, does the LEDs blink?

tonvanhieu, 2020-02-25:
I'm checking it.
I just change device from device list and the "Programming Algorithm" then I can flash it normally. But i have problem with debugging by Keil 5. Variables that i add to WATCH window don't change, it's always ZERO.
Yes, the center LEDs blink.
I have checked I2C, it's same as yours ( I2C_0, PB6 and PB7). Because i can't debug so i don't know if it can read or not ( i will check it by USART later).

EmanuelFeru, 2020-02-25:
And do you get Green LED after the blink? That is an indication the MPU6050 is working fine.

tonvanhieu, 2020-02-25:
I will check it later ( i have to go to school now). 
By the way my MPU  is 6052, is it same with 6050?

Reply edit: yeah i know about that, in my case i use my USART-Bluetooth module (HC-05 - from my old project 😄 ).

EmanuelFeru, 2020-02-25:
MPU6052 is probably a newer variant, theoretically it should work, practically I didn't check because I don't have it.

tonvanhieu, 2020-02-26:
BIG PROBLEM WITH DELAY FUNCTION.
It's always stuck when meet delay. I don't know why.  When i delete introDemoLED(100); it configuring MPU non-stop.
[image]

EmanuelFeru, 2020-02-26:
You need to power the board with at least 12V. I noticed similar funny behavior when I was powering the board with 7.5V from an external source.
So, yes, either use the an external source with 12V or use the mainboard power.

tonvanhieu, 2020-02-26:
delay_ms(); make program be stuck. How to fix it :((. i feel so impotent.
I have tried power it by mainboard but nothing change.
Sensors work well but mpu not yet.

tonvanhieu, 2020-02-26:
It's all about delay function. when i change delay time in mpu_init() below 10 the board can go 1 more step (writing DMP). So, if i can find out what happen to delay_ms() i can solve that. But i still can't :((. Delay_ms() kill me.
[image]

EmanuelFeru, 2020-02-26:
How did you unlock your GD32 chip? 

I had similar behavior with one board. To fix it I did the following (please try):
* Lock back the GD32 with ST Link Utility
* Unlock the GD32 as described [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-Flash) 
* At Step 6, check all the boxes for "User configuration option bytes"
* Flash and try

tonvanhieu, 2020-02-26:
I follow you at step 6 and this box appear
[image]
Any wrong with it?

EmanuelFeru, 2020-02-26:
Hahaha...:) Good that it works. The error is no issue.
Yeah.. sometimes the unlocking is not done properly and you get funny behavior.

closed 2020-02-26 by tonvanhieu

Nassimraz, 2024-12-18:
Hello my board with gd32 turned off after unlocking mcu ( I didn't hold the power button while unlocking)