EFeru/hoverboard-sideboard-hack-GD #5: Wrong Euler Angle.

tonvanhieu, 2020-03-01:
I saw that Euler angles that calculate from DMP are so wrong.

When i checked with 3D Visualization Demo, the angle rotated more about 8 times than reality.

I saw your clip upload on Telegram, it's so cool and smooth.
 
Have you ever meet that behavior? Please help me teacher.

I have sent you a short clip on telegram, please accept my request (my telegram: Tan Nguyen: +84859818681)

EmanuelFeru, 2020-03-01:
If you check with Debug variant, the angles are OK. The visualization needs more work.. 
The Serial read in the visualization is not that great. It is not that easy to read out pure integers from serial because is in Java. The visualization would work better if we send text separated by a space or other delimiter. This can be more easily proceeded by the tool.

Now that you mentioned I think is better to use the Vi dualization with debug variant, send 'e' and read out the angles.

tonvanhieu, 2020-03-01:
> If you check with Debug variant, the angles are OK. The visualization needs more work..

You are right, the angles in Debug variant is much better. I'll will try again with the debug variant as you say.

> I could not see your video. Is better to send me a private message directly in telegram.

I dont't know why but my account cant send message to another contact. Maybe a mistake. I have sent Telegram a complaint.

EmanuelFeru, 2020-03-01:
> I dont't know why but my account cant send message to another contact. Maybe a mistake. I have sent Telegram a complaint.

I use the app on my phone and PC. They both work well.

closed 2020-03-01 by tonvanhieu

EmanuelFeru, 2020-03-01:
@tonvanhieu , I fixed it .. works perfectly now :) I will commit the sketch in 15min with other stuff too.