EFeru/hoverboard-sideboard-hack-GD #7: Arduino control

pav18, 2020-04-28:
Hello.
Would you share a code to control this board via UART?
Thank you.

EmanuelFeru, 2020-05-02:
It is there already. You just need to send a structure of the following format via UART:
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/c703b309300446bea40bd06d833f206ab9da93ba/Src/main.c#L44-L57

Make sure you send the structure at the Baud rate and Start frame as specified here:
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/c703b309300446bea40bd06d833f206ab9da93ba/Inc/config.h#L69-L70

Of course you can modify the structure to your needs.

closed 2020-05-21 by EmanuelFeru

pav18, 2020-05-23:
It looks like feedback structure but how to control sideboard's leds etc?

EmanuelFeru, 2020-05-23:
In the structure, there is `cmdLed`. With this you control the Leds. Each bit represents one Led on (1) and off (0).