EFeru/hoverboard-sideboard-hack-GD #8: Can't flash firmware

jusubbi, 2020-09-23:
I tried with keul uvision5 and st-link software. Keul gives error about missing packet (i installed GD32F130 package). St-link detect chip as some st version and start to flash it, but stops at beginning. I can disable read out protection.

Do i have to install some button to put chip in programmin mode like mainboard (must hold power button while flashing)?

EmanuelFeru, 2020-09-23:
For sideboard, you don't need to press any button, just make sure the MCU gets power, either via the ST link or mainboard. I prefer to shut down the mainboard and have power only from ST link during programming.
Can you make a picture of you sideboard? 

Usually locking unlocking the chip twice helps in solving the issue.
Also in another thread was mentioned that GD sometimes lie about the flash memory size. So there is a chance that even that you chip supposed to have the datasheet flash size it can be that it has lower.

jusubbi, 2020-09-23:
I'm using just sideboard without connection to mainboard and power comes from st-link.
Also locking / unlocking gives error, but still doing it?
Here is pictures from board and errors:

[https://photos.app.goo.gl/91fUEPshA6DXv26W6](url)

EmanuelFeru, 2020-09-23:
Looks exactly as my board.
Does the project in Keil compile without errors? And can you flash via Keil?

jusubbi, 2020-09-24:
Keil compile works without errors, but flash doesn't.

[image]

[image]

jusubbi, 2020-09-24:
I guess these are fake chips. I flashed demo firmware (27916 bytes) and everything goes fine.

 
[image]

Then i tried bigger file (10208 bytes) and it fails again.

[image]

EmanuelFeru, 2020-09-24:
Wow. Then what is the real size of flash? 8 kB :-o

jusubbi, 2020-09-25:
Is there way to check out real flash size? I changed address to 0x07007000 and now it flash it, but it's a mess. I don't need IMU or leds, just those 2 sensors working as buttons, so is it possible to reduce firmware size to make it work with these faulty chips?

[image]

Candas1, 2020-10-04:
If you need only buttons there is an option to use right sensor cable on the mainboard with buttons. Wouldn't it be good enough? https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/2890745fbb5879b52821231b53c217f886344f4b/Inc/config.h#L250

EmanuelFeru, 2020-10-04:
Can you update the firmware of your ST Link v2 programmer to the last version? For me it flashes and I remember I did than. I haven't tried before, with the shipped ST link v2 firmware.

See here how to do it:
https://youtu.be/QXbw82Jdtw8

jusubbi, 2020-10-04:
> If you need only buttons there is an option to use right sensor cable on the mainboard with buttons. Wouldn't it be good enough? https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/2890745fbb5879b52821231b53c217f886344f4b/Inc/config.h#L250

Yes, that be enought. I just really need to know how to connect buttons. Are they between rx/tx or rx/tx to ground?

jusubbi, 2020-10-04:
> Can you update the firmware of your ST Link v2 programmer to the last version? For me it flashes and I remember I did than. I haven't tried before, with the shipped ST link v2 firmware.
> 
> See here how to do it:
> https://youtu.be/QXbw82Jdtw8

I have latest version V2.J34.S7

Candas1, 2020-10-04:
> Yes, that be enought. I just really need to know how to connect buttons. Are they between rx/tx or rx/tx to ground?

I think it should be connected to 3.3v. Which variant are you using? Want do you want to do with the buttons?

jusubbi, 2020-10-04:
> I think it should be connected to 3.3v. Which variant are you using? Want do you want to do with the buttons?

rx or tx to 3.3v via switch?

I'm using hovercar variant, so i want use 1 button for phase advance on/off and another to change motor driving mode.

Candas1, 2020-10-04:
Yes. 
This is not yet possible out of the box, but I have it on my to-do list.
Do you want to implement it? 
You would have to copy this https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/2890745fbb5879b52821231b53c217f886344f4b/Src/util.c#L1184
And use button1 and button2 variables as input instead. It could be run after read_command function in main loop.

jusubbi, 2020-10-04:
I'm not a programmer, so i don't know where i have to copy those lines? I'm better with hardware.

If this is too hard, i think is better for me to just compile firmware with different settings until i found best combination.

Now i have "stock" variant_hovercar with just calibrated ADC values and battery + field weakening enabled.

Candas1, 2020-10-04:
OK forget about it then.
Regarding sideboard, if code size is the problem and you only need buttons, you can deactivate mpu sensor here
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/423126eab5739a976c11d7179601a56c5a8f3d52/Inc/config.h#L45

jusubbi, 2020-10-04:
> OK forget about it then.
> Regarding sideboard, if code size is the problem and you only need buttons, you can deactivate mpu sensor here
> https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/423126eab5739a976c11d7179601a56c5a8f3d52/Inc/config.h#L45

I guess that firmware size is the problem. If i just comment that line 45 does it reduce filee size?

Candas1, 2020-10-04:
Yes it should

jusubbi, 2020-10-04:
Still too large file :( i can flash 2kt to this chip. Now this new file with 6050 disabled is 6kt

Candas1, 2020-10-04:
2k seems really low
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/423126eab5739a976c11d7179601a56c5a8f3d52/GD32F130C6T_FLASH.ld#L21
This mcu is supposed to have 32k of flash

jusubbi, 2020-10-04:
> 2k seems really low
> https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/423126eab5739a976c11d7179601a56c5a8f3d52/GD32F130C6T_FLASH.ld#L21
> 
> 
> This mcu is supposed to have 32k of flash
> Do you have same concern with the other sideboard?

Yes i know. Same problem with another sideboard too.

jusubbi, 2020-10-06:
I now tested those BLDC control types with flashing modified firmware to mainboard and only working type in my use is FOC TORQUE. Also field weakening tested and i don't wanna use it, because my vehicle goes too fast (so it illegal to drive) and use 100% more power just to speed up +33%. So this issue can be closed.

closed 2020-10-06 by jusubbi

redeemarr, 2022-09-15:
I've had the same issues with sideboard gd32f130c6t6 chip: st-link utility fails to flash relatively big files.
However [st-flash](https://github.com/stlink-org/stlink) worked flawlessly (I used windows binaries). And st-link utility was able to read flashed data. Seems like those 'broken' chips are not so broken.
I haven't tested it yet, though.

AlfreddGco, 2023-11-27:
@redeemarr you are amazing! That just made it work and finally got to upload all the firmware to the chip. Thank you! I have seen several forums with this exact issue, will try to share your answer!