EFeru/hoverboard-sideboard-hack-GD #9: Different SWD Programming Pins

timtimmahh, 2020-11-20:
On my hoverboard sideboards the SWD programming pins differ from the ones specified in [sideboard pinout](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/master/docs/pictures/sideboard_pinout.png). From top -> bottom mine go:

- VDD 3.3V
- PA14 / SWCLK
- PA13 / SWDIO
- GND

It took a while to figure out since I thought there was something wrong with my STLINK/V2 programmer, so it might help others with the same board to add a notice in the readme for that.

EmanuelFeru, 2020-11-20:
Thank you for the info. I will as it to readme. How does your GD board looks like?

timtimmahh, 2020-11-20:
It looks almost identical to the one in your image I guess it just has some minor pin changes on the SWD interface. I know the main board connector is the same, but I haven't checked the other pins yet.
[image]
[image]

mamert, 2021-03-12:
I can add a photo of two more boards that use this pinout:
[image]

Thing is, while they can be unlocked (throwing an error, but it works after) and erased, flashing fails - similar to "[flashing fails](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/issues/8)" issue, maybe?
The resolution for that issue seems to be "no longer care", but it might be good to know how to identify those that WON'T work...

[image]
[image]

EmanuelFeru, 2021-03-12:
@mamert  thanks for the input. For you which board works and which one not?

mamert, 2021-03-12:
Both behave identically (fail).

Though I have not had a single success with the sideboards,
I have built hoverboard-firmware-hack-FOC successfully in the same VSCode,
so the environment is OK

perdmen, 2021-12-26:
Hi! I have a similar situation, but it's important to me.

SWD
- VDD 3.3V
- PA14 / SWCLK
- PA13 / SWDIO
- [GND](url)

First I clicked Option Bytes and selected disabled, got an error message "Failed to set option bytes! Please reset the target and try again", but the firmware has been erased, and the Erase Chip is running successfully.
when I click Program & Verify and want to install the firmware I get an error: "Programming error @ 0x8000000!"

[image]

[image]

[image]

captFuture, 2022-03-21:
I have the same issue here with a YST board - no upload possible (and i checked the chip 5 times if it is a GD32F130 ...) - interestingly it shows as a :STM32F10xx Medium-density in ST-Link.

Has anybody sorted this out yet?

Candas1, 2022-03-21:
WDG_SW option should be set.
Pinout can differ between sideboards, you need to check which pin is connected to vdd, swdio, swclk and gnd. 
The chip can also be defective.

redeemarr, 2022-09-15:
Just in case, I was [able](https://github.com/EFeru/hoverboard-sideboard-hack-GD/issues/8#issuecomment-1247440703) to flash it with [st-flash](https://github.com/stlink-org/stlink)

derFrickler, 2022-11-16:
Can you explain how you did the flashing via st-flash? 
cant' get mine flashend as well. Mins is the green boad with the purple connector shown above.

redeemarr, 2022-11-16:
@derFrickler, nevermind, I was wrong.
st-flash flashes the chip with no errors, but it doesn't work.
Make sure you checked nRST_STOP, nRST_STDBY and WDG_SW checkboxes while flashing with ST-Link utility. This was the reason it didn't work in my case.
https://github.com/EFeru/hoverboard-sideboard-hack-GD/issues/25

derFrickler, 2022-11-16:
Great! seems that did the trick.
Thanks a lot