EFeru/hoverboard-sideboard-hack-GD #12: Zero value and IMU failed

GMahmoud, 2021-01-24:
Hey @EmanuelFeru,
I followed the all the steps to get sideboard data (GD32F130C6T6 with  MPU-6050A).
I unlocked the board and got it flashed. However, when I am monitoring Euler value I am getting only zeros.
```
Command = e
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
```
When I run a self-test, I am getting this result

```
accel: 319802219 -323885 134236467
gyro: -1743067344 1087314292 -1256236507
Gyro failed
Accel failed
Compass failed
``` 
Any idea what's causing this ? 
Thanks for your help 
Cheers

EmanuelFeru, 2021-01-29:
Strage, so you have gyro and accel but not DMP data. We need to debug further a bit, can you display the DEBUG output when you first power on the sideboard. It should say `Configuring...writing DMP.. Ok`
And maybe place some `printf(..)` in the Euler angle calculation to see what happens.

GMahmoud, 2021-01-30:
Hello @EmanuelFeru, 
I confirm I am getting the `Ok` message after the configuration (see attached logs)

However, I looked a bit in your code to add some printfs. I did not focus on euler angle only as these are computed from quaternions which are zeros as well. I added some log to  [`dmp_read_fifo`](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/master/Src/mpu6050_dmp.c#L1240) function that computes quaternions and it's returning -1 because of `mpu_read_fifo_stream`.
In this function I found that variable  [`fifo_count`](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/master/Src/mpu6050.c#L1761) is usually equal to 0 as well as `tmp` array (see debug printf in the logs)

Could you help me please to debug `i2c_read` function ? 

```
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Configuring MPU6050...  writing DMP...  OK
=== HELP ===
h: Print Help
x: Print Serial AUX
8: Set Accelerometer on/off
9: Set Gyroscope on/off
r: Print Registers
a: Print Accelerometer
g: Print Gyroscope
q: Print Quaternion
e: Print Euler angles in deg*100
t: Print Temperature in degC*100
p: Print Pedometer
0: Reset Pedometer
1: Set DMP/MPU freq 10 Hz
2: Set DMP/MPU freq 50 Hz
3: Set DMP/MPU freq 100 Hz
,: Set DMP interrupt to gesture
.: Set DMP interrupt to continuous
f: Set DMP on/off
v: Set Quaternion on/off
w: Test low-power accel mode
s: Run self-test (device must be facing up or down)
============
DEBUG: fifo_count: 0, length: 32, tmp[0]: 0, tmp[1]: 0
Command = e
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Roll:0 Pitch:0 Yaw:0
Command = e
Command = q
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
qW:0 qX:0 qY:0 qZ:0
Command = q
```

GMahmoud, 2021-01-30:
Other question: Do you have any idae what is the difference between `MPU-6050` and `MPU-6050A` ?

EmanuelFeru, 2021-01-30:
Not sure about the difference, but it could be a reason why is the DMP and angles not working. Because the quaternion is zero for you.
I am thinking.. could it be that the DMP firmware is different. I think we need to investigate it.

2com68, 2021-02-08:
Hello. I have a question about the hoverboard-sideboard-hack-GD project.
I plan to use your firmware to upgrade a small electric car. With wheels and a gyro-scooter motherboard.
To run two wheels, do I have to install the firmware on the two side boards? 
Or do you need to connect one side board from both sides to the motherboard?
I need the central motherboard to be able to be controlled by an adult via a car model receiver. And to be able to control the pedal with a hall sensor. 
Will this firmware be able to implement such management?

EmanuelFeru, 2021-02-08:
Are you sure you have the same sideboards? Because I have a feeling that you have gen2 boards. If yes, it will not work with this firmware you have to use https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2
This firmware is for the sideboards of gen1 boards, where the 2 motors are driven from [one mainboard](https://raw.githubusercontent.com/EmanuelFeru/hoverboard-firmware-hack-FOC/master/docs/pictures/mainboard_pinout.png)

2com68, 2021-02-09:
> Вы уверены, что у вас такие же серванты? Потому что у меня такое ощущение, что у вас платы Gen2. Если да, она не будет работать с этой прошивкой, вы должны использовать https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2
> Эта прошивка предназначена для боковых панелей плат Gen1, где 2 двигателя приводятся в действие с [одной материнской платы.](https://raw.githubusercontent.com/EmanuelFeru/hoverboard-firmware-hack-FOC/master/docs/pictures/mainboard_pinout.png)

Hello. I have a Gen1 (photo attached) where two motors are powered by a single motherboard.
For my purpose, I need control via the receiver and the RC remote control (photo attached). And also to be able to switch to the control from the pedal (photo attached). Pedal with hall sensor.
Здравствуйте. У меня Gen1 (фото прикрепил), где два двигателя приводятся в действие одной материнской платой.
Для моей цели нужно управление через приемник и пульт RC (фото прикрепил). А так же что бы была возможность переключаться на управление с педали (фото прикрепил). Педаль с датчиком холла. 

[image]
 [image]
[image]
[image]

2com68, 2021-02-09:
> Здравствуйте. У меня есть Gen1 (фото прилагается), где два двигателя питаются от одной материнской платы.
> Для моих целей мне нужно управление через ресивер и пульт дистанционного управления (фото прилагается). А также иметь возможность переключаться на управление с педали (фото прилагается). Педаль с датчиком холла.
> Здравствуйте. У меня Gen1 (фото прикрепил), где два двигателя приводятся в действие одной материнской платой.
> Для моей цели нужно управление через приемник и пульт RC (фото прикрепил). А так же что бы была возможность переключаться на управление с педали (фото прикрепил). Педаль с датчиком холла.

Подскажите, какие я должен совместить прошивки для получения такого управления? 
Я не говорю на английском, перевожу через Google c Русского на Английский. Перевод может некорректным. 
Скажите если что-то будет не понятно в вопросе. Попробую сформулировать иначе и добавить картинок. 
Tell me, what should I combine the firmware to get such control?
I do not speak English, I translate through Google from Russian to English. The translation may be incorrect.
Tell me if something is unclear in the question. I will try to formulate it differently and add pictures.

EmanuelFeru, 2021-02-09:
@2com68 seems like you have the normal supported board. From what I understand you need the Hovercar variant. This works with using the mainboard and one sideboard connected to the Right (short cable).
**Schematic:**
- https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR

**Mainboard firmware:**
- Uncomment VARIANT_HOVERCAR in platformio.ini, build and flash

**Sideboard firmware:**
- Uncomment VARIANT_HOVERCAR in platformio.ini, build and flash

2com68, 2021-02-11:
Hello. Thank you for the information. I will study it. I hope I'll figure it out by the summer.
Just got a question. The process of turning on the reverse can only be done by double pressing the brake pedal? Or can I turn off the brake pedal? Do the reverse through the switch?

EmanuelFeru, 2021-02-12:
I didn't test myself, but theory says the reverse should engage with a switch by double pressing as replacement to Brake pedal.
However, if you want to be sure you can implement a very easy if else based on the switch.

2com68, 2021-02-19:
> Я не проверял себя, но теория гласит, что задний ход должен включаться переключателем двойным нажатием вместо педали тормоза.
> Однако, если вы хотите быть уверены, что можете реализовать очень простой if else на основе переключателя.

Hello. I didn't quite understand what to do. Tell me please. Need to edit something in the firmware? To implement the reverse through the switch? The photo shows an example of a switch. Or should I plug in a switch instead of a pedal, without changing anything in the firmware?
[image]
[image]
[image]

Candas1, 2021-02-26:
> Other question: Do you have any idea what the difference between `MPU-6050` and `MPU-6050A` is ?

Hi, I also have a some boards with mpu6050a, maybe we can collaborate. Have you made any progress in the meantime? We can chat if you are in the telegram group

GMahmoud, 2021-02-27:
@Candas1 With pleasure, I was quite busy lately to debug more about this issue. 
A question: Did you get the same problem with your board ? 
I am not in the telegram group, but we could chat in it if you could please explain to me how to join 😸 

Cheers

Candas1, 2021-02-27:
Yes.
All seems OK but I don't get DMP data. 
If I disable DMP in config.h I can at least see accelerometer and gyro data. 

Here is the link to the hoverboard telegram group:
https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw

My nickname is Candas there

2com68, 2021-03-12:
> @ 2com68 похоже у вас нормальная поддерживаемая плата. Насколько я понимаю, вам нужен вариант Hovercar. Это работает с использованием материнской платы и одной боковой панели, подключенной справа (короткий кабель).
> **Схема:**
> 
> * https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR
> 
> **Прошивка материнской платы:**
> 
> * Раскомментируйте VARIANT_HOVERCAR в platformio.ini, соберите и прошейте
> 
> **Прошивка серванта:**
> 
> * Раскомментируйте VARIANT_HOVERCAR в platformio.ini, соберите и прошейте

Hello. I downloaded the firmware from the specified link.
As I understand it, the automatic calibration of the handles is disabled by default?
Tell me what changes I should make to enable automatic calibration.
At what point do I need to make changes to connect the side board and get i-bus control?
I built the firmware without making any changes. I installed the firmware in the motherboard and side board. The motherboard is turned on, but you can't turn on auto-calibration mode.

pop-nick, 2025-03-30:
> Hey @EmanuelFeru, I followed all the steps to get sideboard data (GD32F130C6T6 with MPU-6050A). I unlocked the board and got it flashed. However, when I monitored Euler value I got only zeros.
> 
> ```
> Command = e
> Roll:0 Pitch:0 Yaw:0
> Roll:0 Pitch:0 Yaw:0
> Roll:0 Pitch:0 Yaw:0
> Roll:0 Pitch:0 Yaw:0
> Roll:0 Pitch:0 Yaw:0
> Roll:0 Pitch:0 Yaw:0
> ```
> 
> When I run a self-test, I get this result
> 
> ```
> accel: 319802219 -323885 134236467
> gyro: -1743067344 1087314292 -1256236507
> Gyro failed
> Accel failed
> Compass failed
> ```
> 
> Any idea what's causing this ? Thanks for your help Cheers 😃
> 
> PS: After the flash, I have a steady green LED light.

Have the same issure with Gen 1. In my case the sensor is 6050A.

Candas1, 2025-03-30:
It won't work with 6050A
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Sideboards