EFeru/hoverboard-sideboard-hack-GD #13: 'FLASH' overflowed

mamert, 2021-02-21:
Hi, I'm obviously missing something since make fails with errors.

Help, please?

Ubuntu 16.04
stlink installed from stlink-1.6.1-1_amd64.deb
VARIANT_DEBUG selected
Makefile: added -std=gnu99 (else I get "'for' loop initial declarations are only allowed in C99 or C11 mode")

Result: 
(...)
arm-none-eabi-gcc build/gd32f1x0_adc.o build/gd32f1x0_can.o build/gd32f1x0_cec.o build/gd32f1x0_cmp.o build/gd32f1x0_crc.o build/gd32f1x0_dac.o build/gd32f1x0_dbg.o build/gd32f1x0_dma.o build/gd32f1x0_exti.o build/gd32f1x0_fmc.o build/gd32f1x0_fwdgt.o build/gd32f1x0_gpio.o build/gd32f1x0_i2c.o build/gd32f1x0_ivref.o build/gd32f1x0_misc.o build/gd32f1x0_opa.o build/gd32f1x0_pmu.o build/gd32f1x0_rcu.o build/gd32f1x0_rtc.o build/gd32f1x0_slcd.o build/gd32f1x0_spi.o build/gd32f1x0_syscfg.o build/gd32f1x0_timer.o build/gd32f1x0_tsi.o build/gd32f1x0_usart.o build/gd32f1x0_wwdgt.o build/system_gd32f1x0.o build/main.o build/util.o build/systick.o build/mpu6050.o build/mpu6050_dmp.o build/setup.o build/i2c_it.o build/gd32f1x0_it.o build/startup_gd32f1x0.o -mcpu=cortex-m3 -mthumb   -specs=nano.specs -TGD32F130C6T_FLASH.ld  -lc -lm -lnosys  -Wl,-Map=build/firmware.map,--cref -Wl,--gc-sections -o build/firmware.elf
(...)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7-m/libc_nano.a(lib_a-errno.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: build/firmware.elf section `.rodata' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 684 bytes
collect2: error: ld returned 1 exit status

By the way, EmanuelFeru/hoverboard-firmware-hack-FOC (the mainboard) builds successfully.

EmanuelFeru, 2021-02-21:
Hi,
Variant_Debug is quite large. I build and flash with Platformio or Keil and there I get like 98% memory.
I never flashed with Ubuntu so it could overflow.  Have you tried to set optimizations flags? Maybe is reduces code size.
Otherwise you could disable the MPU sensor by commenting out MPU_SENSOR_ENABLE in config.h (in case you don't need it) because it saves a lot of space.

mamert, 2021-02-21:
@EmanuelFeru Thank you for the response.
I don't know of any optimization flags aside from `-Os`
and that is already the most space-saving there is.
Setting `DEBUG = 0` seems to have made zero difference: "region `FLASH' overflowed by 684 bytes"
Not touching MPU_SENSOR_ENABLE since MPU6050 is the main thing I'm interested in

I am guessing that something in my build chain differes from yours, causing this size difference...

VARIANT_HOVERBOARD and VARIANT_HOVERCAR both build correctly.
Can they be used with the 3D demo, too?

EmanuelFeru, 2021-02-21:
The demo works with the Debug variant. Can you try building  with Platformio and see if you succeed?
I will need need to update the code in Processing because in the last updates the Serial output was changed so most probably 3d demo does not work at the moment because it does not read the angles properly. I will check.

mamert, 2021-02-21:
Thank you.
Platformio: can't test anytime soon; threw me into Python dependency hell on my old Ubuntu.
I'll try on another machine next week

mamert, 2021-02-22:
Platformio build of VARIANT_DEBUG succeeds on Windows

Good enough for me, but what to do with this issue (make on ubuntu -> overflow)?
Leave open or close?

EmanuelFeru, 2021-02-22:
If you are ok with building in Platformio, you can close it. However, in case you want to check what can be done so it builds under Ubuntu, then we can keep it open to see what can be done.

For the Processing code to run we need to update these:
* baud rate to be updated to `115200` here:
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/53ae2e5f151f0937cc8114a89eedd81fbe624026/docs/sketch_processing/sketch_processing.pde#L19
* and Serial delimiter data needs to be updated here
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/53ae2e5f151f0937cc8114a89eedd81fbe624026/docs/sketch_processing/sketch_processing.pde#L82-L88
to read the Serial data that is now received like this `Roll:10 Pitch:20 Yaw:30`, (before it used to be `Euler[RPY]: 10 20 30`)
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/53ae2e5f151f0937cc8114a89eedd81fbe624026/Src/mpu6050.c#L3731-L3732

The output was updated so it works nicely with the tool from Candas https://candas1.github.io/Hoverboard-Web-Serial-Control/
However, in case you want o fix the Processing sketch feel free to create a Pull Request, I am ready to accept it :-)

EmanuelFeru, 2021-02-22:
For the Processing Serial read you can inspire from @Candas1 tool code. He does the parsing like this:
https://github.com/Candas1/Hoverboard-Web-Serial-Control/blob/6a52d7515cc6cf3dddb422e67a384d6e35fe19e5/js/serial.js#L309-L330

mamert, 2021-02-22:
Thank you for the code and links!

closed 2021-02-22 by mamert