EFeru/hoverboard-sideboard-hack-GD #14: No data received via Main Serial

victorfdezc, 2021-03-11:
Hello, I am having some problems in getting data from the IMU with an Arduino via Serial. First I have to say that im new to this kind of projects, so I apologize if can't explain properly.
I have a sideboard with a GD32F130 C6T6 and a MPU-6050A. I have unlocked the micro and then flashed it (with lot of difficulties because most of the time I get errors meanwhile writing the pages). Before flashing, I modified config.h to have the Hoverboard Variant because it includes the Serial Control via serial port (I have also tested with Serial Feedback deactivated).
Then, I modified [this Arduino program](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Arduino/hoverserial/hoverserial.ino) to define the same [Feedback struct](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/master/Inc/util.h#L28) that is sent via serial by the Sideboard.
And finally, I connected the 3.3V and GND pins from the SWD programming pins to power the board, and the GND and RX (pin number 2) pins from Arduino have been connected to GND and TX from the connector that connects the Mainboard with the Sideboard.
However, Arduino does not receive anything via serial. In fact, I have enabled DEBUG_RX in the Arduino program and I only receive 0s.
I have also tested to disable DMP as said in [this comment](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/issues/12#issuecomment-787069858), but again nothing happens.
I don't know what is wrong and I don't have more ideas. Any help would be appreciated.
Thanks

EmanuelFeru, 2021-03-12:
We know that there is an issue running the DMP on MPU-6050A, running on MPU-6050 is ok. However i think your issue is in the Serial communication as I read your comment.
Are you sure the sideboard is sending data? With an FDTI adapter or an oscilloscope you can check that by connecting it directly to the Tx pin of sideboard.
Then next step is to check the Arduino code.

Candas1, 2021-03-12:
Hi,

I couldn't get DMP to work with MPU-6050A.
Only if I disable DMP in config.h, I can get Accelerometer and Gyroscope raw values in DEBUG MODE

Candas1, 2021-03-12:
Couldn't post a message on invensense forum, I guess it was moderated, need to try again.
[This](https://github.com/liuheng135/glvio/blob/c5969897c3c10cbb06281a60b03720a663e05d27/glvio-vision/drivers/i2c/device/mpu6050.c#L132) is the only hint I found about possible differences between MPU6050 and MPU6050A, but still I didn't get it to work

victorfdezc, 2021-03-14:
Thanks for the response! I have purchased an FTDI adapter and it seems that the problem is the serial communication. I have tried multiple times to upload the Debug Variant but it always fails before flashing completely. Disabling the MPU allows me to flash the Debug Variant, but again, nothing is sent :(