EFeru/hoverboard-sideboard-hack-GD #15: GD32F103C8T6 vs GD32F130C6T6

sdegeorgio, 2021-05-25:
Hi Emanuel,

I have two sets of sideboards one of each shown below.  One set has GD32F103C8T6 and the other set has GD32F130C6T6.

Programming the GD32F130C6T6 fails usually after writing somewhere between 4 and 12 pages and this is consistent for both sideboards in the pair. They unlock and erase / erase verify ok. I think there was another issue already open where people had experienced the same problem but no solution yet.

Programming the GD32F103C8T6 boards always succeeds, however the code does not appear to run properly, LED 4 or 5 turns on and nothing further happens, no communication with main board. Have you ever successfully run on this MCU?

Do you have any experience with either of these issues or any idea where to start to look to debug?

[image]
[image]

Candas1, 2021-06-30:
Hi, 

Sorry I just saw that issue now:
For the 130C6T6 board, can you try to set WDG_SW in [option bytes](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-Flash) ? 

For the 103C8t6, can you try the STM fork ?

sdegeorgio, 2021-07-01:
Wow, thank you so much!  Your solution for each respective sideboard worked!

I searched the web for so long before opening the ticket and you solved it so quickly.

Stephan.

closed 2021-07-01 by sdegeorgio