EFeru/hoverboard-sideboard-hack-GD #18: region `FLASH' overflowed by 44 bytes

davidrib, 2021-11-28:
Could someone help-me here?? Exceeds 44 bites :(((((((
not strong programmer
[image]

davidrib, 2021-11-28:
My chip version is GD32F130C6 and I have more two boards with GD32F130C8, not tested yet but should work they have more memory, even so, only because 44 Bites is not possible adapt the firmware to fit in 32K ?? I'm not a big programmer but culd someone help?

davidrib, 2021-11-28:
Success shortened some console messages
[image]
I think the project  could be updated 
Thanks

EFeru, 2021-11-28:
Maybe we could use a define to enable/disable all Debug messages.

davidrib, 2021-11-28:
> Maybe we could use a define to enable/disable all Debug messages.

I'm not a advanced programmer is the first time I use platforio, but I'm wondering if the compiled version have or not info to start/stop add breackpoints  to some debugger that could be disabled? Should save a lot of space?
(Sorry my bad English)

davidrib, 2021-11-29:
Done! 35% free space!

[image]

davidrib, 2021-11-29:
> Maybe we could use a define to enable/disable all Debug messages.

I'm so 🥇 dumb 🥇  :((((((((

Sorry

EFeru, 2021-11-29:
https://github.com/EFeru/hoverboard-sideboard-hack-GD/blob/50fa92b606b3ecde737a5f6263beb0faeec166a9/platformio.ini#L47

This just defines the VARIANT_DEBUG macro which is used through out the code. If you comment it out, you will not have debug pieces of code in the compilation. So, it won't be the VARIANT_DEBUG anymore.

I just updated travis build and I also get 44 byte overflow. I think the new platformio or better said new librarier from platformio use more resources... it happened before.. I will see what can be done soon.

maxgerhardt, 2021-12-21:
44 bytes oversize is really on the edge. I know it built before with an earlier version of my platform-gd32. So, if new features aren't needed and rather, stability is needed, I can do a PR for the `platformio.ini` to use fixed git commit hashes that results in a good build. 

(But then possible new features in platform-gd32 can't be taken advantage of)

maxgerhardt, 2021-12-23:
I've proposed a fix for it in https://github.com/EFeru/hoverboard-sideboard-hack-GD/pull/19.

PR #19 (2022-01-01): Fix versions of GD32 platform and SPL package
Restores successful compilation for the VARIANT_DEBUG environment which in the current version overflows by 44 bytes for some reason.

```
RAM:   [=         ]  14.9% (used 612 bytes from 4096 bytes)
Flash: [==========]  98.8% (used 32364 bytes from 32768 bytes)
Building .pio\build\VARIANT_DEBUG\firmware.bin
====================[SUCCESS] Took 2.53 seconds ====================
```

This patch has the disadvantage that further developments in the GD32 platform are not taken into this project -- the version is frozen now, but also stable in regards to  [... cut]
closed 2022-01-01 by EFeru