EFeru/hoverboard-sideboard-hack-GD #20: Help with this board

krisstakos, 2022-04-04:
So I've cracked an old hoverboard containing that pcb. It uses GD32F103C6T6 (reports 32k but actually it's less I think 8k)
And I was able to compile & flash the DEBUG example without the MPU_SENSOR definition
[image]

Any help of the pinout of where is that UART that I can hook and observe/send commands?
One additional question - with that VARIANT_DEBUG can I send commands to move the motor? and what I should use if that's not the right variant (not interested in MPU sensor just sending command to move in specified direction and speed)

Candas1, 2022-04-05:
Hi, 

You are posting in the wrong repository, this firmware is for sideboards, those have no motor control.
Please check [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility)

krisstakos, 2022-04-05:
Hi, you are right it's for sideboards(with integrated mosfets). But the board chip is GD32F103C6T6 and I didn't find any reference there. It looks like to me that's a mix between the two. Hardware wise is 90% like in the url you mentioned but as I said there is no c6 version is it possible to flash the c8?

Candas1, 2022-04-05:
Well the wiki is made to point you in the right direction, your need to look for help in those repositories or in the telegram group.

Are you sure you are reading the chip model well ?

krisstakos, 2022-04-05:
Hi, I'm 100% sure it's GD32F130**C6**T6 (previously wrote it 103 by mistake)

Candas1, 2022-04-05:
That's my point
You are asking for help about the wrong chip in the wrong repository 🤣
Have you seen [this](https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2/issues/13) ?

krisstakos, 2022-04-05:
That looks promising I guess. Thanks!

closed 2022-04-05 by krisstakos