EFeru/hoverboard-sideboard-hack-GD #21: Firmware.bin Will Not Flash

1BadAz, 2022-04-05:
I tried forking the code and when I try and upload firmware.bin it will not download, ST-Link is saying "warning core is locked up." It will download firmware.elf (idk what that is) and when I power the board it beeps in sets of three. I am attempting to use US_ARtT variant. When I hook it to an Arduino it seems to have no effect.

krisstakos, 2022-04-05:
You have ti unlock your chip first: https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-flash

1BadAz, 2022-04-06:
I already disabled read-out protection. I have locked and unlocked it a dozen times still no change. I also did follow the step in that link to check all of the user config options. There is still no change.

1BadAz, 2022-04-06:
Now saying"Error: Mass erase operation failed. Please verify flash protection.

closed 2022-05-12 by 1BadAz