EFeru/hoverboard-sideboard-hack-GD #22: debug variant not compiles

Ami603, 2022-05-05:
Hello, i'm trying to compile a debug variant, however i get the following error:

/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: build/firmware.elf section `.rodata' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 2696 bytes

so it seems it's bigger than flash region on the GD chip.

Candas1, 2022-05-05:
Hi,

This is known issue, I believe with the current toolchain the footprint is bigger.
What chip do you have on your board ? If it's a C8T6 instead of a C6T6, you can increase RAM from 4K to 8K and FLASH from 32K to 64K [here](https://github.com/Candas1/hoverboard-sideboard-hack-GD/blob/0f3f84751a42dcdba8c0009c450f6eee44611336/GD32F130C6T_FLASH.ld#L20) as a workaround.

Ami603, 2022-05-05:
C6T6 here,  but i managed to compile the code to a suitable firmware by commenting the last while(1) loop since it seems to be outside of the debug loop.

Ami603, 2022-05-05:
I've just noticed my own fault by commenting the last loop, there's no chance to input any options back.

Candas1, 2022-05-06:
What are you trying to achieve ?

Candas1, 2022-05-06:
I was trying to replace some of the math functions for euler angle processing by the ones from qfplib on my [fork](https://github.com/Candas1/hoverboard-sideboard-hack-GD)

Can you please test it ?

@EFeru, I could create a pull request if it works well

Ami603, 2022-05-06:
Just toying with some robotics using hoverboard components, i was trying to obtain the values from the MPU via serial,to gain some knowledge.
Main idea is to have something like https://github.com/LuSeKa/HoverBot, but instead making use of the original hardware from the Hoverboard.

Candas1, 2022-05-06:
I don't have wiki edit access in this repository, but I had created [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Sideboards)

Ami603, 2022-05-06:
Thanks, so now i can see i have four sideboards with GD32F130C6T6 and MPU6050A, and two sideboards with same GD32 and MPU6050.

Candas1, 2022-05-06:
It should work with the mpu6050's

Candas1, 2022-05-17:
Have you had a chance to try my fork that compiles ?

Ami603, 2022-06-05:
Not yet, somehow i've given up on using the sideboards for now, have yet to find free time to resume the tests.