EFeru/hoverboard-sideboard-hack-GD #24: MPU init and config intermittent failure

vortex314, 2022-08-27:
Hi,
sometimes the mu_init() and mpu_config() fail.  When I power cycle the board it's good again for some time. 

- So I was wondering is there a software method to reset the mpu6050 ?  Not sure if the I2C is failing or the MPU6050. 
- I don't find anymore a detailed schematic ?  Somebody knows a location of the pdf ? The old link doen't work anymore. 

Any help is appreciated,
Lieven

Candas1, 2022-08-29:
Hi,

Which link ? The link might be broken but the there are many pdfs in the docs folder.
I didn't use those boards too much but I never encountered the issue you mention about.
Maybe some `delay_1ms()` here and there might help.

vortex314, 2022-09-01:
@Candas1  thanks for the feedback. Concerning the link to the schematic, my bad, I was mixing up with another repo. But a link to the full schematic of the sideboard would be useful. 
Also I didn't find any hardware reset pin for the mpu6050.,except power-on /off.

Candas1, 2022-09-01:
I believe there is no such schematic

vortex314, 2022-09-01:
I noticed that also my raspberry pi on the same 5v power supply was complaining about under voltage, so this is probably the cause and generating brown out.

closed 2022-09-01 by vortex314