EFeru/hoverboard-sideboard-hack-GD #30: New Sideboard layout found with same processor

eroboticdude, 2023-10-31:
I have a new variant of sideboard with 2 led headers and a slightly changed pinout. I will update with the pinout I find.
[image]

So far, a lot of the pins line up for the sensors and battery meter pins. One notable difference is that this board is fed +36V instead of +15V, so the LEDs are most likely a different voltage config. There is also 3 colors in the front lights, so they are potentially RGB. will update!

eroboticdude, 2023-10-31:
Here is an update on the pinout:
[image]

eroboticdude, 2023-11-01:
Here's another update: I've discovered the PB10 header for some reason fried the power supply/regulator on the far left side of the board and now the board doesn't boot from +36v in. 
[image]

I first thought it was the regulators that were bad, but that is not the case after all. It turns out that one of the transistors near the capacitors on the left side has gone bad from what I can tell. The 2 regulators are rated for 12v at 0.5A or 1A at 3.3v so 300mA of led current should not have been the issue. Nonetheless It is totally fried and so are some of the output transistors, not sure what is happening.

eroboticdude, 2023-11-09:
Here is the final Pinout:
[image]
The 2 LED Headers are RGB, but I can only seem to get the left header working. PD0 and PD1 don't seem to function properly in my modified firmware.

DDex89, 2026-09-09:
Here another layout for you guys:

[image]

Also pinout of the mcu from rep. has minor errors with PC 13,14,15 ports marking. here fixed one:

[image]